jkelleyrtp/dw1000-rs #61: Docs fail to build

jamesmunns, 2018-10-23:
```
error: `[Frame::write]` cannot be resolved, ignoring it...
  --> /home/james/.cargo/git/checkouts/ieee-802.15.4-2c4cb3c4e568cea2/6d2059f/src/mac.rs:74:12
   |
74 | /// Tells [`Frame::write`] whether to write the footer
   |            ^^^^^^^^^^^^^^ cannot be resolved, ignoring
   |
note: lint level defined here
  --> rust-dw1000/src/lib.rs:7:9
   |
7  | #![deny(warnings)]
   |         ^^^^^^^^
   = note: #[deny(intra_doc_link_resolution_failure)] implied by #[deny(warnings)]
   = help: to escape `[` and `]` characters, just add '\' before them like `\[` or `\]`

error: `[Ping]` cannot be resolved, ignoring it...
  --> rust-dw1000/src/ranging/mod.rs:16:43
   |
16 | //! 2. Anchors regularly send out pings ([`Ping`]).
   |                                           ^^^^^^ cannot be resolved, ignoring
   |
   = help: to escape `[` and `]` characters, just add '\' before them like `\[` or `\]`

error: `[Request]` cannot be resolved, ignoring it...
  --> rust-dw1000/src/ranging/mod.rs:18:10
   |
18 | //!    ([`Request`]) for each ping they receive.
   |          ^^^^^^^^^ cannot be resolved, ignoring
   |
   = help: to escape `[` and `]` characters, just add '\' before them like `\[` or `\]`

error: `[Response]` cannot be resolved, ignoring it...
  --> rust-dw1000/src/ranging/mod.rs:20:19
   |
20 | //!    response ([`Response`]).
   |                   ^^^^^^^^^^ cannot be resolved, ignoring
   |
   = help: to escape `[` and `]` characters, just add '\' before them like `\[` or `\]`

error: Could not document `dw1000`.
```

Probably worth adding `cargo doc` to the CI.

hannobraun, 2018-10-23:
Thanks for the report!

I'll do a documentation pass before the first crates.io release.

hannobraun, 2019-02-11:
Some observations:
- The failure was probably due to `#[deny(warnings)]`, which is no longer enabled. Now we just get warnings.
- Some of these come from the `ieee802154` crate. Those warnings a) make no sense, b) don't show up when running `cargo doc` directly for that crate, and c) only appear on nightly. I've found issues of false positives for `#[warn(intra_doc_link_resolution_failure)]`, so I'm guessing it's that.
- Warnings are still forbidden in the CI build, so adding `cargo doc` there is not possible as long as we get false positives.

closed 2019-02-11 by hannobraun

hannobraun, 2019-02-13:
For the record, I figured out why those `ieee802154` doc warnings show up (because we're re-exporting one of its modules). See #71.

commit 824e5530 (2019-02-11): Add missing links in API documentation

Close #61