jkelleyrtp/dw1000-rs #63: Ranging examples don't seem to work in release mode

jamesmunns, 2018-11-18:
With a fairly recent nightly:

```text
active toolchain
----------------

nightly-x86_64-unknown-linux-gnu (default)
rustc 1.32.0-nightly (6e9b84296 2018-11-10)
```

I am unable to get the examples working in `--release` builds. Normal builds work correctly, and I see the output.

I have a sneaking suspicion that https://github.com/nrf-rs/nrf52-hal/issues/33 is related, however I have tried this using the `beta.4` release of `nrf52832-hal`, which contains fixes for this.

I also sometimes get errors that look like this via semihosting on the tag side: `panicked at 'Failed to initialize DW1000: Spi(Transmit)', libcore/result.rs:1009:5`

jamesmunns, 2018-11-18:
Whoops. I had multiple copies of the dependencies locally. When *actually* using the changes in https://github.com/nrf-rs/nrf52-hal/pull/34, this now works in release mode.

Sorry for the noise, but I will keep this open until https://github.com/nrf-rs/nrf52-hal/pull/34 has landed upstream, and we have fixed our dependencies. This will probably be fixed naturally by landing https://github.com/braun-robotics/rust-dw1000/pull/62.

hannobraun, 2019-01-18:
Both https://github.com/nrf-rs/nrf52-hal/pull/34 and https://github.com/braun-robotics/rust-dw1000/pull/62 have landed. The only thing required to close this issue for good would be to update the minimum version of the `nrf52832-hal` dependency to `0.6.1`.

hannobraun, 2019-02-11:
Now that this crate no longer depends on `nrf52832-hal`, this issue no longer belongs here. Opened https://github.com/braun-robotics/rust-dwm1001/issues/71.

closed 2019-02-11 by hannobraun