jkelleyrtp/dw1000-rs #88: Stuck in block!

thejpster, 2019-04-26:
During the workshop, we saw that when lots of packets were being sent (by another user drawing to the screen in https://github.com/ferrous-systems/embedded-trainings/blob/master/beginner/docs/segment-4.md), we saw several instances of programs locking up inside the block! macro.

One of the attendees expanded out the block macro and it seems the rx_future.wait() call just returns nb::WouldBlock forever (or at least, for several minutes). This seems like a bug? Is it possible it's missing some edge trigger from the radio? I haven't looked in to what the comms looks like so I'm just guessing here.

hannobraun, 2019-04-28:
Thank you, @thejpster. I'll look into it when I'm back home (or maybe I'll get a chance during the impl days).

hannobraun, 2019-05-28:
Finally managed to take a look at this. Sorry for the delay.

I can reproduce the problem using the following setup:
- One DWM1001 board running the [`dw1000_only_tx`](https://github.com/braun-embedded/rust-dwm1001/blob/f37cd929889ea745f24fd9618420c9f62ae33130/examples/dw1000_only_tx.rs) example.
- Another board running a stripped down version of the [`dw1000_only_rx`](https://github.com/braun-embedded/rust-dwm1001/blob/f37cd929889ea745f24fd9618420c9f62ae33130/examples/dw1000_only_rx.rs) example. I removed the debug output and the LED blinking loop, only blinking once when receiving successfully. This results in more messages being received, which helps trigger this problem more quickly.

Observations:
- Sending messages (on the board that ends up freezing) is not required to trigger the problem. So my first suspicion, that this is related to switching between the RX/TX states, was unfounded.
- When starting a new receive operation, the problem disappears and the driver goes on to operate normally.

This means there's a simple workaround: Never use `block!` to wait for messages, always use a reasonable timeout. For example using [`embedded-timeout-macros`](https://crates.io/crates/embedded-timeout-macros), as demonstrated in the [DWM1001 examples](https://github.com/braun-embedded/rust-dwm1001/tree/master/examples).

I'll keep investigating to find out what causes the problem.

hannobraun, 2019-05-29:
Okay, I think I've got it. Every time it freezes, a flag is set that indicates a frame was automatically rejected for not matching the receiver's address. I was under the impression that frame filtering was handled automatically by the hardware, and that I don't have to specially handle this flag.

This seems partially true, but sometimes a filtered frame seems to bubble up and stop the ongoing receive operation, which leads to the buggy behavior. If this is correct, the simple fix for this issue is to check that flag and report it as an error.

I'm currently running a test to verify my fix. So far everything looks good, but I'd like to observe it for a while longer.

hannobraun, 2019-05-29:
My test has been humming along for more than an hour now, which is far longer than it ever took me to reproduce the issue.

I've opened a [pull request](https://github.com/braun-embedded/rust-dw1000/pull/92) which I'll merge as soon as CI passes.

closed 2019-05-29 by hannobraun

thejpster, 2019-05-29:
Yay! Thanks for taking the time to look at this :)

commit 9a487625 (2019-05-29): Handle frame filtering rejection

Close #88