jkelleyrtp/dw1000-rs #94: FYI `SYS_STATE` is now available in the datasheet

korken89, 2019-07-09:
Hi Hanno!

I just wanted to make you aware that in the latest version of the DW1000 register datasheet the  `SYS_STATE` register is finally specified with all the internal states of the chip.
This could be used to make the low-level driver more complete. :)

BR Emil

hannobraun, 2019-07-09:
Oh, nice! I'm already using it for some things (don't remember what exactly), based on info I found in some secondary source, but having it properly documented is of course much better :-)

korken89, 2019-07-09:
Indeed! 
A colleague and I was looking today to start porting our C++ code to Rust and had a look at the DW1000 library and saw this addition in the datasheet. :)

PR #97 (2019-08-08): Complete definition of SYS_STATE register
Close #94
commit e1d5d2ab (2019-08-08): Complete definition of SYS_STATE register

Close #94
closed 2019-08-08 by hannobraun