jkelleyrtp/dw1000-rs #118: Timeout error in ranging tag example

hannobraun, 2020-03-27:
This happens from time to time:
```
panicked at 'Failed to send ranging request: Timeout', examples/dw1000_ranging_tag.rs:136:13
```

I've observed this in both release and debug mode.

hannobraun, 2020-03-30:
I can no longer reproduce this and can't find a reason in the code for why this would happen. I'm leaving this be for now. If anyone else sees this or has more information, please post here!

jonathanpallant, 2020-04-16:
I'm now seeing this:

```
panicked at 'Failed to send ranging request: Timeout', examples\dw1000_ranging_tag.rs:136:13
```

Sometimes rebooting the chip helps.

jonathanpallant, 2020-04-16:
Also, because of panic-semihosting, if you don't have semihosting enabled in your debugger, you won't know why it has stopped running.

When using J-Link GDB, add "monitor semihosting enable" to the GDB file when you change the port to :2331.

hannobraun, 2020-04-17:
Thanks for the report, @jonathanpallant! I don't have time to look into it right now (and couldn't reproduce it last time I did), so any help in figuring out why this happens is appreciated.

What's weird about it, is that I've never seen this problem on the anchor. At least I've never experienced that an anchor stops working. So either the anchor is ignoring this error (would be worth checking), or it doesn't happen there.

I've suspected that something is wrong with the timer, not the driver, but when I checked, I didn't find any obvious misuse of the timer, nor was there any indication of bugs in the HAL.

hannobraun, 2021-12-14:
All examples work as of #145. I'm assuming this issue has been fixed too, whatever it was. There have also been a few updates of the HAL dependency since then.

If it turns out to still exist, we can reopen this issue, or open a new one.

closed 2021-12-14 by hannobraun