jkelleyrtp/dw1000-rs #126: Publish new cargo crate version?

smklein, 2021-01-24:
After getting set up with this crate - downloading the most recent version published to crates.io (0.4.0), I got an application set up, and tried to copy the "blinky" LED example as a basic check that things worked.

Unfortunately, the compiler complained - even the simple rust blinky example broke, because of a version mismatch: the version on crates.io still depends on nrf52832-hal version 0.8.1, even though that's been fixed on master to point to a more recent revision.

Would it be possible to publish a version 0.5.0 of this crate? That would make the crates.io documentation stay in sync with the repo. I've also seen folks using git tags to identify "which revision of the source code + examples are consistent with what can be downloaded from crates.io"

hannobraun, 2021-01-26:
Thanks for opening this issue, @smklein. A new release is definitely overdue. I've added it to my task list. Hopefully I'll get to it some time this week.

hannobraun, 2021-01-27:
New versions are out!

closed 2021-01-27 by hannobraun