jkelleyrtp/dw1000-rs #144: Version bump?

jkelleyrtp, 2021-11-30:
Would it be possible to bump the published version so that the GitHub repo and published crates version are in sync?

hannobraun, 2021-12-02:
Hey @jkelleyrtp, thanks for opening this issue.

I don't want to publish a new version while #140 is completely unaddressed. I haven't found time to address it yet, and don't see that changing soon.

Ideally, someone would fix the examples. But failing that, I would be fine if someone could confirm that the driver itself is working correctly. I don't want to publish a broken release, but without the examples in a working state, I can't confirm that the driver itself is working.

If you have a project using this driver, and can confirm that the latest state from this repository is working, I'd be happy to publish a release.

jkelleyrtp, 2021-12-02:
I can go through and make sure the examples still work. From what I can tell, no, the ranging examples don't work. I haven't tried any of the other examples (beyond blink). I think #134 might've broken a few things.

hannobraun, 2021-12-02:
Quite possible, but I actually think the examples were broken before that. Although I might misremember.

Those ranging examples (and to a lesser extent, the other radio examples) have been a huge pain for as long as this project existed. I forgot all the details, but I fixed those a few times; I think mostly because of timing issues.

It's possible that the driver works, but the examples don't. If that were the case, it wouldn't be great, but at least I'd feel comfortable publishing a new release.

Kethku, 2021-12-02:
Yeah note I wasn't able to get the project working before or after. But I haven't dug into what could be going wrong

jkelleyrtp, 2021-12-03:
I've verified that these examples do work:
- [x] blink.rs
- [ ] dw1000_delayed_tx.rs
- [ ] dw1000_id.rs
- [x] dw1000_only_rx.rs
- [x] dw1000_only_tx.rs
- [ ] dw1000_ranging_anchor.rs
- [ ] dw1000_ranging_tag.rs
- [ ] dw1000_reg_modify.rs
- [ ] dw1000_reg_rw.rs
- [ ] dw1000_reg_short.rs
- [x] dw1000_rx_tx.rs
- [ ] dw1000_subreg.rs
- [ ] dw1000_subreg_ext_addr.rs
- [ ] lis2dh12.rs
- [ ] uarte.rs
- [ ] uarte_heapless_string.rs

I should have a PR coming soon that upgrades all the dependencies for both crates, moves the examples to defmt, and ensures all the examples work properly.

hannobraun, 2021-12-03:
Thank you, @jkelleyrtp, this is great work!

hannobraun, 2021-12-14:
Addressed in #150.

closed 2021-12-14 by hannobraun