jkelleyrtp/dw1000-rs #146: Bug: ranging on battery panics in compute_distance

jkelleyrtp, 2021-12-03:
I haven't figured out *why*, but when I pull my tag off USB and leave it on its own power, the calculations in compute_distance fail and panic.

The error happens on this line:
```rust
    let time_of_flight = (rtt_product - rt_product) / sum;
```

Apparently, the rt_product becomes larger than rtt_product (not sure how) and the subtraction fails.

On USB. `(rtt-rt) = 4875522386940746496 - 4875520436757001728 = 1,950,183,744,768`
````
INFO
        Ping reply time: 2179770156
        Ping round trip time: 2179753344
        Request reply time: 2236713088
        Request round trip time: 2236731234

└─ dw1000_ranging_basestation::__cortex_m_rt_main @ examples/dw1000_ranging_basestation.rs:230
INFO
            rtt_product: 4875522386940746496
            rt_product: 4875520436757001728
            rt_sum: 4416483244
            rtt_sum: 4416484578
            sum: 8832967822

└─ dw1000_ranging_basestation::__cortex_m_rt_main @ examples/dw1000_ranging_basestation.rs:256
```

On battery: ( 4811215469605391972 - 4811217180826287836) = -1,711,220,895,864
```
INFO
        Ping reply time: 2174124223
        Ping round trip time: 2174103004
        Request reply time: 2212944932
        Request round trip time: 2212965743

└─ dw1000_ranging_basestation::__cortex_m_rt_main @ examples/dw1000_ranging_basestation.rs:230
INFO
            rtt_product: 4811215469605391972
            rt_product: 4811217180826287836
            rt_sum: 4387069155
            rtt_sum: 4387068747
            sum: 8774137902
```

My guess is that my batteries are really old and the tag briefly browns out leading to unusual numbers.

hannobraun, 2021-12-03:
Oh man, what a bug :sweat_smile: 

I don't have lots of experience with these kinds of things, so I can't really comment. What I do know is, that hardware can be crazy, so your guess sounds plausible in that regards.

I've seen brown out detection functionality on STM32 chips. Maybe there's something similar on the nRF?

hannobraun, 2021-12-14:
@jkelleyrtp It looks like this has been addressed with the additional error variant added in #145? If so, this issue can probably be closed.

jkelleyrtp, 2021-12-14:
Yes. The problem is still here, but seems to be an issue that project authors will need to take care of.

closed 2021-12-14 by jkelleyrtp