jkelleyrtp/dw1000-rs #152: Looking for new maintainer

hannobraun, 2022-01-12:
Hey folks!

I'm [embarking on new adventures](https://github.com/sponsors/hannobraun) and am currently in the process of reducing my involvement in the Embedded Rust community, so I can focus on that. As part of this process, I'm stepping down as maintainer of this repository.

I am willing to transfer this repository (and the crates on crates.io) to a new maintainer. If anyone is interested in that, please let me know by posting here or [contacting me privately](https://hanno.braun-odw.eu/contact/).

cc @jamesmunns (who has write access to this repository)
cc @jkelleyrtp (who has been contributing recently)

jkelleyrtp, 2022-01-12:
Hey - I love the new journey you're embarking on.

I've been dreaming about an auto-CAM system for a while (either through pure heuristics or some level of ML) so what you're doing is totally inline with my thinking. Seems like we share quite a few interests :)

I'd be happy to maintain or co-maintain this crate since my involvement with the DW series is ramping up.

hannobraun, 2022-01-12:
> Hey - I love the new journey you're embarking on.
>
> I've been dreaming about an auto-CAM system for a while (either through pure heuristics or some level of ML) so what you're doing is totally inline with my thinking. Seems like we share quite a few interests :)

Yeah, seems like we do!

I'm also interested in extending my CAD work towards CAM/slicing, but one thing at a time. Getting the CAD part to work alright will already be enough work :smile: 

Maybe we should jump on a call some time to talk about that stuff. It'll probably be a bit of a pain, timezone-wise, but surely worth it.

> I'd be happy to maintain or co-maintain this crate since my involvement with the DW series is ramping up.

That's great, thank you! Do you want to wait for a bit, to see if anyone else is interested, or do you want me to transfer the repository right away? Both options are fine with me.

jkelleyrtp, 2022-01-13:
> Maybe we should jump on a call some time to talk about that stuff. It'll probably be a bit of a pain, timezone-wise, but surely worth it.

For sure! I don't mind putting up with timezone issues :)

> That's great, thank you! Do you want to wait for a bit, to see if anyone else is interested, or do you want me to transfer the repository right away? Both options are fine with me.

I'd be happy to take care of things right now, and then leave this issue pinned in case anyone else wants to jump in.

hannobraun, 2022-01-13:
> I'd be happy to take care of things right now, and then leave this issue pinned in case anyone else wants to jump in.

Looks like I can't transfer the repository to you, directly. I've invited you as a collaborator with admin privileges. Once you accept, you should be able to transfer the repository to your own namespace.

> For sure! I don't mind putting up with timezone issues :)

Great! Just [drop me an email](mailto:hanno@braun-odw.eu) or [hit me up on Matrix](https://matrix.to/#/@hanno:braun-odw.eu), then we can arrange something.

hannobraun, 2022-01-14:
I forget to transfer the crates on crates.io. Sent invites for [`dw1000`](https://crates.io/crates/dw1000) and [`dwm1001`](https://crates.io/crates/dwm1001) now.

closed 2022-02-05 by jkelleyrtp